FrSky-OS/HORUS-X10-X10S-X12S-bug-report #1: [Hardware] Frameloss / holds with 1.2.18 IXJT firmware

RealTadango, 2016-10-16:
I use X8R and X4R receivers and experience strange failsafes. I watched the s.bus data and see about 4% of frameloss constantly. I almost crashed a jet today during a 1 second failsafe period! Telemetry was working and logged the RSSI at 50+ during the failsafe. 

I use OpenTX as OS but others have the same issues with FrSky OS. 

Some my main question: Is the 4% frameloss normal??? It is any distance and constantly.

projectkk2glider, 2016-10-17:
There is a lot of speculation and reports on forums which indicate that Horus has some kind of RF link issue. One example http://www.rcgroups.com/forums/showpost.php?p=35977416&postcount=620 (this thread has most reports)

The nature of the reports suggest that it is not a simple link loss scenario (low RSSI) but something else. It seem that RF link between IXJT and receiver is OK (at least on a signal level), but the model still experiences a failsafe event. The occurrence of this is very random and non-repeatable.

Based on the reports I have concluded the following:
 * the issue **does not happen with Taranis** (even with the same receivers) 
 * the issue only happens on Horus when using internal IXJT module (so far nobody has reported what happens if external XJT module is used)
 * the issue happens on both **internal and external antenna** (both using IXJT)
 * the issue happens **regardless of the OS used** on the radio

So it seem that either IXJT or the radio itself is the culprit.

The IXJT module is the obvious target for this issue. The only authority here is FrSky and it is up the them to try to debug this issue. 

The other target is the radio itself (the main board). According to the reports the issue happens on both OpenTX and FrSky OS. Despite of that we can not rule out the OS, because most probably both systems use very similar or possibly equivalent drivers to talk to the IXJT module.

Basically the issue can happen at two levels:
 * the radio OS sends bad data to the IXJT module
 * the IXJT module receives good data, but sends out garbage or even has a temporary lockup.

Using external module will not rule out either suspect. The proper test case to hunt down this issue would be:
 * log the SBUS signal on the receiver
 * log the IXJT internals: the received stream, sent packets, etc....
 * setup the radio with the sawtooth signal on one of the channels.
 * an issue would be detected when the SBUS signal is either: 
  * interrupted
  * has failsafe flag set
  * the sawtooth waveform on dedicated channel is interrupted.
 * when the issue is detected the other logs at the same time-stamp should reveal the source of the problem.

In reality only FrSky can do the above testing, because the internals of IXJT module are proprietary.

RealTadango, 2016-10-17:
Perfect description. The main thing i noticed is the very often frameloss flag on the S.Bus signal. It is not with a constant interval but the percentage of lost frames is very constant. I did however notice during a test of maybe 30 minutes the frameloss started at 3.5% and rose to 5%. So many lost frames (distance of 0,2 to 3 meters has the same percentage) is not normal behaviour. I a normal flight maybe 100 framelosses in 10 minutes are to be expected max. That would be 0.33% max, in flight. This must be an indicator of the underlying issue and i hope anyone can confirm frameloss with the IXJT module and not with an external module or a Taranis.

bryan39-2, 2016-10-17:
4-5% SBUS Frame Loss

Very interesting data Tadango. How is this measured? Do you have a
logger on the RX SBUS output?

I don't know how an IXJT handles what I presume is SBUS data from
FrOS/OpenTX but if it is asynchronous this might result in regular frame
loss. It could also be, for a similar reason, the data logger.

A certain loss of packets is to be expected in any shared spectrum
environment but what matters for us is the rate of occurrence, any
clustering behaviour like with tape recording dropouts, and what
criteria are used to trigger fail safe.

Does anyone know how many channels XJT hops over, how many channels
there are in total, the hops per second and the number of payload
packets (8 servo channels?) per hop?

I have observed club mates, turn up and turn TX on for a whole session
without regard to what effect it might have on people actually using
their TXs. At some level, this must increase spectrum congestion and
the risk of loss of data packets.

JapieG, 2016-10-17:
https://youtu.be/YRAeK53mNOM
Here you can see the Horus hopping. 

When recording all the hops you see the Horus using its full spectrum.

RealTadango, 2016-10-17:
I use custom logger on the S.Bus port and check Every frame for the frameloss flag.

Adelaxie, 2016-10-18:
Hi, all. Thanks for your disscusion on the issue but there is no differences between IXJT and XJT on this factor. We will try to reproduce the issue you reported and find out the cause if it does exist.

RealTadango, 2016-10-18:
Same RF chip, different supporting hardware, different firmware. I will compare with an external XJT module as soon as it arrives.

RealTadango, 2016-10-18:
This is the most detailed description of a failsafe i have so far:
I had the Horus sitting on a table (wood) about 3-4 meters from my plane. I have 2 receivers in the plane (a X8R and a X4R) with only one transmitting telemetry (X4R). The receivers are connected to a Powerbox Cockpit SRS through S.Bus. I had another X8R receiver without telemetry bound to the transmitter and had attached a simple S.Bus frameloss blinker (Arduino that blinks the led everytime a S.Bus frame with the frameloss flag set has been detected). I wanted to test if the framelosses are the same for every receiver. The green led dims a bit when there is a S.Bus frame with the frameloss flag. This happens for all 3 receivers at exactly the same time. I was watching the receivers when suddenly all 3 receivers showed a failsafe (green led off, red led on) for about 1 second. All S.Bus frames at that moment had the frameloss flag set. The powerbox registered the failsafe also for both receivers. The transmitter did not sounded the telemetry lost alarm or anything else. RSSI was high 80, sometimes 90 before and after. 

This happened in flight also with another plane (jet) and another X8R receiver. It also happened before when setting up a new plane (glider) with also a different X8R. Untill now i have seen this failsafe 3 times without any reason (the flying plane was close with good rssi levels throughout the failsafe).

Next time i will try to log every S.Bus frame for a long period (about 10 hours) for detailed analysis of the framelosses.

bryan39-2, 2016-10-18:
@RealTadango Interesting but you cannot rule out a common source of interference (an external influence) that affects all three RXs. They are in close proximity. 

There are four sets of firmware involved: the OpenTX, the IXJT TX, the XJT in the RX and the arduino one in your logger.  You could rule out the OpenTX and arduino if you could get your monitor to log what OpenTX is passing to the IXJT. There should be no bad frames period.  I am assuming that OpenTX uses SBUS to talk to the IXJT.

RealTadango, 2016-10-18:
I want to rule out OpenTX because people are having failsafes with FrSky OS to. Both with FCC and EU LBT firmware. I want to ruleout the RX part also because the same setup has been flying perfectly for others with a Taranis for a long time and all failsafes start (even at very close range where range / power is clearly not an issue) when they started using the Horus. All reports have one thing in common: Sudden failsafe without a clear reason. That is why i am focussing on the failsafes. I noticed the many framelosses when watching the raw S.Bus data and wonder if it is related. I don't know it other system mask it more then FrSky but i think it is strange there are so many. Both with FCC and LBT firmware.

The IXJT module is controlled by PXX. See https://github.com/opentx/opentx/blob/next/radio/src/pulses/pxx_arm.cpp for the source. An external XJT module is controled by PXX also which looks like a FrSky own serial format. You could use some unused S.Bus flags for that but why? The IXJT has a antenna switch also which is controlled over the PXX stream.

RealTadango, 2016-10-19:
I did a big comparison between the internal IXJT module and a new XJT module and have a conclusion: The framelosses / failsafes are NOT worse with the internal module. During my test i even see more framelosses with the external module. I captured 1.000.000 frames from the modules to compare both in percentage framelosses, failsafes and more then 5 sequential framelosses.

The environment has much influence on frameloss but still does not explain the failsafe moment for which 100 or more frames must be lost. That is a lot of lost data for a long period.

projectkk2glider, 2016-10-19:
@RealTadango are you sure the device you are using to detect frame loss is working correctly? We don't want to jump to some false conclusions based on wrong measurement results.

It would be interesting to see the results of your device with the known good combo - ie Taranis radio. This would also show if your testing methodology is valid.

RealTadango, 2016-10-19:
Yes i requested a Taranis from a friend but i don't have it yet. That would be an interesting test.

My frameloss indication works great. The S.Bus protocol is very simple AND the frameloss flag is set everytime the green led dims. The green led indicates a good frame and blinks every 9msec (or stays on). When a packet is lost the green led dims. With more losses you can see it flicker much more. My powerbox registers differently but they won't confirm how they measure it. I can't understand why.....

I am now testing the XJT through a PPM feed in the Horus and i have not seen a failsafe yet, but it is to early for any conclusion yet. Frameloss percentage is a bit less then with the PXX feed but that varied also. Sequential framelosses of 5 and more are still seen and more then with the internal module.

I will test the XJT with another PPM body that has worked great the past few years with a OpenLRS module. When i have compared all results we might be able to rule out some things.

@Adelaxie would be nice to have some feedback from the FrSky engineers about their thoughts about the failsafes.....

RealTadango, 2016-10-19:
Average frameloss after 2+ hours of testing each mode:
Internal PXX 4.71%
External PXX 4.91%
External PPM 3.30%

Failsafes:
Internal PXX 1
External PXX 1
External PPM 0

Bulk frame lossess (5 or more sequential):
Internal PXX 8
External PXX 54
External PPM 22

So, nothing conclusive yet although the PPM mode looks like it is resulting is less frameloss
Tadango is online now  Send a private message to Tadango  Find More Posts by Tadango	

See my logging data: http://www.rcgroups.com/forums/showpost.php?p=35995687&postcount=777 (about 3.000.000 frames logged :))

projectkk2glider, 2016-10-19:
Your results make no sense (for now). Did you turn off the internal module when you were testing with the external one?

RealTadango, 2016-10-19:
offcourse i turned one module off ;)

BUT i just put the XJT module in my old JR X11.... much less frameloss. I am recording now....

RealTadango, 2016-10-19:
YES!!!!!! The module is perfect. VERY low frameloss in the JD X11 body, 0.25% and dropping. I plug it in the Horus, PPM mode also and it is back at 5% frame loss.

So first real evidence the the Horus IS causing the framelosses. Now lets see WHAT part of it.

projectkk2glider, 2016-10-19:
My money is on the power supply.

RealTadango, 2016-10-19:
Yes it might be. I will try to check with my scope.

RealTadango, 2016-10-19:
It is RF noise on the module power line. I extended the 3 pins to the external module (signal, VCC and GND) and the frameloss was still the same. Extension was about 30cm. I disconnected the PPM signal and it had no effect. I plugged the module in the JR DSX11 body and frameloss was almost nonexistent. Pretty clear results. 

So now to identify the source of the interference. First up is the LCD (easy to disconnect i think).

RealTadango, 2016-10-19:
Yes it is the LCD... i disconnect it and have no framelosses with the external module plugged in

projectkk2glider, 2016-10-19:
:+1: Good work detective, the workaround is simple, just fly without the LCD :grin:

RealTadango, 2016-10-19:
No, not the LCD. I was to soon to judge. But i have almost zero framelosses nu with the Horus with the back removed. I did fidle with the cables to the module section... bad connection? Soldering looks ok as far as i can see.

projectkk2glider, 2016-10-19:
I am not sure how production model looks, but beta hardware had IXJT on the connector (header). Check that connection.

RealTadango, 2016-10-19:
I did, can't see anything. I can get almost zero framelosses, pick it up and is start to go sky high and if i put i down it keeps high. The JR has about the same but the frameloss goes down when i put it down

RealTadango, 2016-10-19:
I keep seeing a big difference with PPM and XJT for the external module in the Horus. With XJT it is impossible to get 0 framelosses, with PPM it is possible. That difference is also seen in my big test. Why is PPM different in frameloss then XJT? Does that make any sense?

kilrah, 2016-10-19:
Really need to examine the power supply rails closely with a scope to see the ripple. Consumption, or peaks could be slightly different.

RealTadango, 2016-10-19:
I will try a separate battery for the XJT later. If that works iets clear.

RealTadango, 2016-10-21:
I can replicate my failsafe with telemetry perfectly now  The issue is with the PXX stream. When i have the external module running with telemetry i can get the receiver to failsafe while seeing telemetry being received (* blinking) and no telemetry lost warning. Only for a very short failsafe. Longer than about 1,5sec does brake the telemetry stream. I only need to pull the PXX pin (power, gnd and SPort still connected). Kilrah and i are both right. About 50% of the time the receiver stays green and no warning is present. The other 50% the receiver loses the green light and goes into failsafe.

So my new question: Does the XJT module stop transmitting frames when it is in the middle of a command packet over PXX? That could explain the higher framelosses in PXX mode also. If so, there is you source. My test findings support it fully and for me the PXX stream is the source of all evil...... or failsafe for now.

RealTadango, 2016-10-21:
Damn, it explains EVERYTHING i have seen during testing so far:
Difference in PPM vs PXX frameloss. PPM was stable and no failsafe. Frameloss on PXX whas not constant and went up and down. This is when the PXX commands and the frames where overlapping more and less. Bulk framelosses were in the high frameloss times, failsafe also.

Telemetry keeps on working during failsafe. It is not a range / antenna / receiver issue.

Hard to replicate, only during very long testings: PXX cyle and frame cycle must be in sync for this to happen. They might differ most of the time so there is no overlap but then the cycles do match up for a short period and the issue starts.

Assumption:
The PXX and sent frames are overlapping at one point causing many framelosses. If they fully overlap the frames are not sent (correctly?). If this syncs up perfectly all frames are lost. Taranis might send out PXX data slower? That would explain why there is no issue there.

More testing to confirm:
Change the PXX stream to send out PXX with a (diffent) delay to brake the syncing. I am also going to test FrSky OS because i have not seen the issue with FrSky OS myself.

Solution:
More robust XJT / IXJT firmware so frames are still sent during a PXX command. Come on FrSky, please contact me!

Joetrin, 2016-10-22:
@Adelaxie seems like @RealTadango is almost at the solution.  @Adelaxie are you able to connect him with the engineer that designed the IXJT module, please.   It would be great if the new firmware release can accommodate his findings.

Thank you

MIKE-DAILEY, 2016-11-17:
I believe this problem was resolved with 1.2.20.  Is anyone still having problems?

RealTadango, 2016-11-29:
No problems reported in a while, case closed :)

closed 2016-11-29 by RealTadango