FrSky-OS/HORUS-X10-X10S-X12S-bug-report #2: Missing telemetry fields for Redundancy bus -

Webbsolution, 2016-10-26:
The following fields from the Redundancy bus (hence refereed to as RB for the rest of my life) are missing from the Telemetry setup screen in 2.1.18 --

The following RB telemetry fields are seen properly in the Telemetry Setup screen

26/44 RB1Volt - 
27/44 RB1Amp
28/44 RB2Volt
29/44 RB2Amp
30/44 RB1 Consum
31/44 RB2 Consum

The following fields from the Redundancy bus (hence refereed to as RB for the rest of my life) are missing from the Telemetry setup screen in 2.1.18 -- found on page 8 of telemetry while an RB is setup as an active sensor - 
RB1_A

RB1_C
RX1_F
RX1_L
RX1_C
RX_1S

RB2_A

RB2_C
RX2_F
RX2_L
RX2_C
RX_2S

MIKE-DAILEY, 2016-10-26:
All the values need to be logged in the data logging and available in the logic switches.

Adelaxie, 2017-10-30:
Added now.

closed 2017-10-30 by Adelaxie