FrSky-OS/HORUS-X10-X10S-X12S-bug-report #3: Telemetry Issue Altimeter Resolution

tzirancamaro, 2016-11-01:
I want to use high-precision variometer FRSKY FVAS-02 with OS FrTX (horus TX)
This sensor has a resolution of 0.1m
But the telemetry system in the 
FrTX OS (version 1.2.20) does not allow me to use this sensor properly because I have to use a minimum resolution of 1*10 equal to 10m (see attached picture)
I request your help to solve this problem and to use a minimum resolution of 0.1m
Thank you
[image]

MIKE-DAILEY, 2016-11-11:
Also applies to the sensor when used with units of feet.

tzirancamaro, 2016-11-11:
Thanks for your answer, I would like to know if there is any way to use the high precision altimeter within its parameters of 10 centimeters, because currently in the horus operative syatem the lowest allowed range is ten meters, which is too much for practical purposes

MIKE-DAILEY, 2016-11-11:
My comment was not an answer.  I was just adding that the same parameter issue exist with feet as well at meters.  It feet the parameter is in increments of 100 feet and not very useful.

tzirancamaro, 2016-11-11:
Oh, I see, then lets hope the technical team of frsky give us a solution, thank you

closed 2017-11-23 by Adelaxie