FrSky-OS/HORUS-X10-X10S-X12S-bug-report #4: Trim Horus in mode1

Salvatico, 2016-11-04:
First of all I am sorry for my "terrible" English
In my Horus I have the following problems:
(firmware FRSKY 1.2.15 and Beta 1.2.20)
in mode1 Glider normal tail, the rudder trim is not working.
In mode1 Glider V-tail, does not work the elevator trim.
Important note: you need to check on the MONITOR screen or with a receiver and servos because in the main screen trim seems to work.

Another important note: In mode1 Airplane V-tail and normal tail the trims  work regularly.

tomkfly, 2016-11-04:
Are you sure yo meant to close this issue?

Salvatico, 2016-11-04:
Sorry, my English is very approximate, you can forward this issue?
Grazie
Paolo

tomkfly, 2016-11-04:
You should only close an issue when the problem is solved. Closing it makes it look like there is no longer a problem.
Your English is fine. I have seen a lot poorer posted by native speakers.

MIKE-DAILEY, 2016-11-04:
Open!

Salvatico, 2016-11-10:
Solved the problem with the new 1.2.21 release.
I want to thank Tomkfly and Mike-Dailey for the help.
Regards
Paolo

closed 2016-11-10 by Salvatico