FrSky-OS/HORUS-X10-X10S-X12S-bug-report #8: System reset or shutdown

joel-1, 2016-11-13:
Hello,

With FrOS 1.2.21 with a heli setup. If I go to the Pit->rudder menu and select LINE as the curve then go to the selection under the switch entry ( is NOR by default ), select it then try to change the entry, the radio will either reset and come back up with no SD card or internal memory found, or just shut off. For me, this is 100% repeatable.

[image]

Adelaxie, 2016-11-14:
Thanks for your reports, we will fix it asap.

joel-1, 2016-11-14:
I have not tried all other menus, but depending on what you find, you might want to scrub other menu items for the same bug.

joel-1, 2016-11-17:
Another observation on this. To be consistent with other menus, on the left side where it says switch, this should be changeable from switch to flight mode. In flight mode, the pull down for switch should disappear. In switch mode, the NOR ( Idle up 1, idle up 2, idle up 3) pull down option should not be there. In Only in flight mode does the NOR pull down option make sense. The pitch->rudder mix curve should be settable per flight mode.

Adelaxie, 2016-11-25:
I think it has been fixed. Can you check it and close the issue if yes?
@joel-1

joel-1, 2016-11-25:
Seems to work, thanks

closed 2016-11-25 by joel-1