FrSky-OS/HORUS-X10-X10S-X12S-bug-report #10: Beta version problem with Internal Flash

Fires04, 2016-11-15:
Hi, I got problem with our Horus ( beta HW ), it starts show internal flash memory error. Everythink other looks ok. Any suggestion ?
I try:

- reflash
- format

If I connect Horus to PC, I see two massstorages one ( sd card is ok ), but internal flash ( needs be formated, but unable to format)

Firmware version : FrSky-OS v1.2.21
[image]

MIKE-DAILEY, 2016-11-15:
If your system is a beta unit with all original beta hardware there is no Flash Memory drive.  And of course the FrTX OS shows the yellow triangle indicating that the Flash Memory drive is not working.  The beta system only use the SD card.

Fires04, 2016-11-15:
ok thanks for info

closed 2016-11-15 by Fires04

MIKE-DAILEY, 2016-11-15:
You might want to contact your FrSky dealer that provided the beta unit to learn the options that FrSky offered beta testers.

Fires04, 2016-11-15:
I'm beta tester, but I did not find specification of beta radios :)

MIKE-DAILEY, 2016-11-15:
Contact your dealer for the options offered to the beta testers.