FrSky-OS/HORUS-X10-X10S-X12S-bug-report #12: THR CURVE issue

tzirancamaro, 2016-11-16:
As you can see in the photo 1, in the menu THROTTLE CURVE, using the option LINE
I setup a straight line of 40% for the FLIGHT MODE IDLE1
Then, as seen in photo 2, I switched to the EXP option
Finally, as seen in photo 3, I returned to the LINE option, where I found my initial setup erased, which was replaced by a line from -100 to +100
I consider urgent that this problem be corrected, because I was about to crash my helicopter in inverted flight when I lost rotor speed
I also consider that other users might not be so lucky and crash their helicopters, due to the problem pointed out
I'm using firmware 1.2.21 in my Horus, Thank you
Jose
[image]
[image]
[image]

janekx, 2016-11-16:
As i understand the purpose of this setting it is one setting per fmode so you set for each fmode the curve and it is stored.  Changing type of curve change stored data and last set is used.

tzirancamaro, 2016-11-16:
Ok, I thought the main setting was LINE and that the EXP setting was secondary
As well, I forgot to mention I didnt change anything in EXP mode, just look at it
Anyway, I think the system should not be allowed to make changes such as I have indicated, for security reasons

janekx, 2016-11-17:
Why, last visited screen is saved.