FrSky-OS/HORUS-X10-X10S-X12S-bug-report #14: Red LED on, Horus does not respond to any operator action

rdeanchurch, 2016-11-20:
I upgraded to 1.2.21 from 2.15 on a new(?no gimbal access on back cover?)/replacement for my Beta unit Horus.  I setup  a new model on FrTX-OS using several power on, edit, power off cycles. I set the Horus on the table next to me, powered off and continued studying.
After about 1/2 hour the haptic went on/of-on/ offf and the LED came on solid RED.  I tried short and long presses on Power button, entry button, return, page up / down.   There was no response on anything.  Let it set over night...Red LED still on.   Power button pressed and held for over 2 minutes by sweep second hand of clock.  Finally after Mike/Jet-Flyer suggestion, removed back cover and disconnected the battery.   Waited a few seconds, plugged battery back on.  Got a low battery warning.  Powerd off normally and back on, I still had a low battery warning.  Started Horus charging and 6 hours later it is still charging.
It was fully charged before I started creating the new model.  A similar report on RCGroups was posted tonight, Sunday Nov 19th.

The user manual has no description of what a red LED means found in my reading...or what to do about it.

BadRaven, 2016-11-22:
The first charge on my new Production Horus took forever, in fact I became convinced it was not going to end at all!  I cannot give charge time as I had to do other things inc overnight so turned the charger off and unplugged it at least three times before it achieved stop charge.

It was however FAR FAR longer than six hours. I did not do anything with it until it had been charged, so maybe avoided the red "battery exhausted?" LED.

After several cycles left on to below half bars and then charges (typical in use operation) it now appears to be ending charge OK, and for about the same time as my Dev Horus, i.e. SLOW!

octaviojss, 2016-11-23:
So i read in openrc forums that a temporary fix for this is to open the radio and unplug the battery for a minute or so and plug it again. It should fix it untill the next same problem occurs.