FrSky-OS/HORUS-X10-X10S-X12S-bug-report #16: FLVSS telemetry issue

joel-1, 2016-11-24:
FrOS 1.2.21 trying to use 2 FLVSS LiPo sensors, both sensors are from 2016 which is after the last FW update I could find. One sensor is ID2 one sensor is ID4. Only six cells show on the telemetry page. Each sensor works fine by itself. When both sensors are connected with 6S packs only cells 1 - 6 show on the telemetry page. If I remove one of the packs the telemetry screen will momentarily show 0V for all cells then randomly show voltages skipping through numbers from the other pack. if I remove the sensor that has no pack connected the display returns to normal. I verified that the sensors have different IDs using the setting tool.

Adelaxie, 2016-11-25:
IDs  you mean physical ID or group Number?
Or can you tell me how you set the IDs differently?

joel-1, 2016-11-25:
I used the servo channel changer tool to change the physical ID from 02 ( default ) to 04 on one of the sensors.

Adelaxie, 2016-11-26:
All the default group number is 0, you need to change other one to 1 (1-15,not 1) to make two sensors that are the same kind to work.
Now ,they send out data with the same application ID and data covers each others' on the sreen.

joel-1, 2016-11-26:
I used my smart port dashboard to change group ID and still have an issue. My last attempt has one sensor at Phyical ID 2 and Group ID 2 ( bat 1A ), second sensor is Physical ID3 and Group ID 3 bat2 A ). I can get 12 cells to display but it does not seem to behave correctly. If I disconnect bat 1A, celss 1 - 6 go to zero as expected. If I reconnect bat 1A the cells reappear. If I disconnect bat some cells from readout 7-12 disappear and cells randomly appear on 1 - 6. Another issue I notice is I have a cellmin alarm set to repeat every 10 seconds. When I disconnect bat 2A, the alarm is constant and seems out of order. Usually it says cell minimum voltage, warning and repeats every 10 seconds. In bad state is repeats cell minimum voltage sever times in a row ( do delay ), the warning several times in a row ( no delay ).
[image]
[image]
[image]
[image]
[image]
[image]

Adelaxie, 2016-11-28:
@joel-1  Will try on this.

joel-1, 2016-12-06:
I saw in the 1.2.23 release notes this was addressed. I am still seeing the same issue. It is possible that when I unplug a battery it behaves differently than if I just let it run down.

Adelaxie, 2016-12-08:
Maybe I misunderstood your issue. Can you make a video ?

EngelMT, 2016-12-08:
We test it also today with 1.2.23. But to get values for 12S is not possible. We st different IDs. In the moment we have no dashboard in stock to set group IDs. Any idea how we or customers can do it without dashboard? With Horus it is not possible to set group IDs.

Adelaxie, 2016-12-09:
Got the issue. Will fix it. @joel-1

joel-1, 2016-12-09:
Good, thanks. So you do not need me to make a video?

Adelaxie, 2016-12-09:
No need. thanks for your feedbacks.

EngelMT, 2016-12-09:
O.k. hope we get our delivery today incl. FSD. Than I will test and come back.

joel-1, 2016-12-19:
Should this be checked in .24 when it's out?

Adelaxie, 2016-12-20:
yes  @joel-1

EngelMT, 2016-12-20:
Adela, with 1.2.24 is also important to set a group ID or do it work without?

joel-1, 2016-12-20:
Hi,
The display now works properly but there is still an issue. When using celsmin warning, it only works with cells 1-8. If I unplug cells 9-16 I do not get a warning. Do you want to leave this issue here or start a new one?

Joel

Adelaxie, 2016-12-20:
a new one please,long post/

joel-1, 2016-12-21:
Cell display is OK. Closing issue

closed 2016-12-21 by joel-1