FrSky-OS/HORUS-X10-X10S-X12S-bug-report #19: New Sticks

EngelMT, 2016-11-28:
New sticks with switches are not easy to mount. The holes in the Gimbals are to small. 4 cables are still possible. 5 cables actually do not work.

I have drilled the gimbals to 3 mm. Now it works with 4 cables well. 5 cables I do not know in the moment.

Bad is that for the 2-Pos. Switch the LS and RS must put out from the main board.

The buttons do not work either. I assume that these should be connected to the two 2-pin sockets on the main board. But no possibility to use it in the moment. Should be a software issue.

This 2-Pos. switch are used for engine (electric Glider) or retract (Airplane). So it must be possible to use it in Spec-Fun. But not possible. Further no switch warning.

Another big problem is that hardly a customer will get the plugs to the cables. It requires a small crimping pliers.

I would therefore suggest that FrSky supplies the sticks with about 10-15 cm cable and then seperate 10 cm long cables which are ready made with the connectors. Then add some shrink tubing.

The customer then only needs to solder the cables together and push the shrink tube over it.

This is workable.

Then I would wish if all movable cables (sticks and gimbals) were from soft silicone cable. So it is a matter of time until the cables break and there are problems. I have the cables on the plug now coated with rubber to get the movement away from the plug. See photo.

[image]
[image]

EngelMT, 2016-11-28:
Some more information. To add the sticks with switches in the software I use AUX1 (RS) and AUX2 (LS) for the 2-Pos. Switches. It is in airplane and in a new created model.

If you look in the servo monitor you will find CH17 and CH18 for the switches. If I switch CH17 the monitor shows 0 - 79. If I switch CH18 it shows 0 - -100. But the CH 17 goes from 79 to 100!!!!

hrenz, 2016-11-28:
Hy Andres,
die Kabel sind einfach zu dick für 5 Kabel.
Da gibt es aus dem Flugzeugbau sehr dünne Teflonlitzen (PTFE-Litze) die sind Spitzen,
da bringe ich auf 3mm  Bohrung locker 8-10 Teflonlitzen durch.
z.B. AWG32, AWG 28,  AWG26, z.B. bei RS-Online : AWG26, 0,14mm²  7/34 Litzen, 
Aber die 3mm Bohrung ist auch nur  1 Schalter = 3 Litzen 0,25mm² vorgesehen.
Die X9E ist da optimal, da gibt es genügend freie Ports, Digital und Analog
die gibt es einfach bei der Horus nicht, da muss man murksen.

--

EngelMT, 2016-11-28:
Hallo Helmut,

EngelMT, 2016-11-28:
Die Sticks haben ja einen Taster und einen 2-Pos. Schalter. Das sind 4 Kabel. Und wenn man den mit Poti und Taster nimmt, dann sind es sogar 5 Kabel.

janekx, 2016-12-01:
please keep in english, cant read it

EngelMT, 2016-12-01:
Helmut wrote me that the wires are to fat for it. He inform me about special thin wires like AWG32 and so on. I wrote him that I know it but FrSky use fat ones and so it is not easy to mount it. And it is impossible to mount the stick with the Pot because there are 5 wires.

Adelaxie, 2016-12-02:
1. We are trying to find wires that more thin and soft
2. We will provide the wires and connectors with wires like you suggested.
@EngelMT 
Thanks.

EngelMT, 2016-12-02:
Good idea! Thanks!

closed 2016-12-02 by EngelMT