FrSky-OS/HORUS-X10-X10S-X12S-bug-report #22: Flight Modes

EngelMT, 2016-11-28:
Customers and we are missing from the beginning the "Flight Modes" in Airplane and Glider Mode. Please working on it. It is hard to explain why a transmitter like the Horus has it not included.
 @janekx
     
janekx commented 5 days ago
Glider have Fmodes
 @EngelMT
     
EngelMT commented 5 days ago
Oh yes, sorry! But in Airplane it is missing.
 @Woita2095
     
Woita2095 commented 5 days ago
... and I suggest: In Copter to have 6 modes instead of 5 as is now. Contollers for example commonly used APM or Pixhawk need 6 modes.

janekx, 2016-12-01:
Glides need more minimal 7, 10 optimal

Adelaxie, 2016-12-05:
In our todo list but in lower priority as this will changes the structure of firmware now a lot.

EngelMT, 2016-12-05:
O.k.  I understand. But it is one of the most important thinks the customers will have. I understand, of course, that you do not want to take any risk at the moment. But it does not help to offer a "high end" transmitter, which has only a middle class software. If she works. The customers have an expectation to the Horus. Not least of FrSky in the announcements aroused. At the moment the transmitter is only partially fulfilled.

closed 2016-12-05 by EngelMT