FrSky-OS/HORUS-X10-X10S-X12S-bug-report #24: Channel mapping Helicopter swash to PPM

eijowk, 2016-11-29:
I have something strange in my Horus setup.
- FrOS 1.21
- Banggood 4 in 1 multi module running PPM+ mode. Stock firmware.
- An old fashioned TRex 450SE v2 helikopter with 120 dgr swash, Spektrum 6ch receiver
Heli setup HR3

Similar set up Banggood 4in1 module and Spektrum rx with small quad works perfect.

The problem:
Channel mappning on the MONITOR display looks correct
But channels are mapped wrong (different than on MONITOR display) and behaves maybe strange on the Spektrum receiver. I haven't had the possibility to measure PPM output in the module bay connector with oscilloscope, but my guess is that the channel order is wrong there.

I think is that there is a bug for Heli mode HR3 when it is mapped to PPM channels.

Adelaxie, 2016-11-30:
>  I haven't had the possibility to measure PPM output in the module bay connector with oscilloscope, but my guess is that the channel order is wrong there.

We did this but found nothing wrong.
Please provide more information of your condition to help us toreproduce it.

eijowk, 2016-11-30:
I made some more short testing yesterday evening. My findings:
-Channel order and behaviour on trainer output correct, checked with transmitter view in Phoenix simulator.
- Tested again in my helicopter- channel order very strange. 
I can add that I run mode 2. The behaviour on helicopter is a bit like HR3 swash mixing was blended with mode 1.

eijowk, 2016-11-30:
Some more testing this evening.
The Horus behaves correct from this perspective. The root cause seems to be that the Banggod 4 in 1 module is mapped AETR, while DSM2 receivers are typically TEAR mapped. Together with swash mixing on a Helicopter it gives a very strange beahaviour which took a while to figure out. Now changed the channel mapping in Horus to adopt to the 4 in 1 module AETR mapping and everything seems to be OK.

The case can be closed now. I will post this on rcgroups as well since I guess that more people will run in to this strange behavior.

closed 2016-11-30 by eijowk

Adelaxie, 2016-12-01:
@eijowk   Thanks.