FrSky-OS/HORUS-X10-X10S-X12S-bug-report #31: 1.2.22 Transmitter GPS

EngelMT, 2016-11-30:
Does not seem to work. Outdoor even after 10 minutes no display. The display "Dist." completely missing.

janekx, 2016-12-01:
Need to be chatched for 20-30mins we test it

EngelMT, 2016-12-01:
O.k. thanks, will test ist again. But it can not be normal I think!

MIKE-DAILEY, 2016-12-01:
It is much faster finding satellites if you turn off the transmitter RF while it is finding the satellites and then turn the RF back on.

EngelMT, 2016-12-01:
Thanks Mike. Will test it. But as I wrote I think it can not be normal. All GPS systems in cars find the satelite in a few seconds.

EngelMT, 2016-12-01:
Mike, you are right. With RF from, the satellites are found within 2 minutes. Bu I missing the term "DIS" and the value to "DIS". Is it a problem of the 1.2.22?

MIKE-DAILEY, 2016-12-01:
I skipped 1.2.22 because of the model change problem.  I can confirm that the DIS: is on the screen and working in the next build release.

EngelMT, 2016-12-01:
O.k. thanks

Adelaxie, 2016-12-02:
The RF module is too near to the GPS so we recommend to turn off the RF module when searching satelites.
DIS should be showed as we test.
Thanks Mike for the answer.

EngelMT, 2016-12-02:
o.k. thanks. Then this can be closed

closed 2016-12-02 by EngelMT