FrSky-OS/HORUS-X10-X10S-X12S-bug-report #36: S6R Bug reports and Change/Enhancement Requests

mpjf01, 2016-12-02:
Is there a separate repository on Github for issues with the S6R and other FRSKY equipment that we could have a link to please?

Adelaxie, 2016-12-15:
For other products,please write mails to our technical support 
sales4tech@gmail.com

closed 2016-12-15 by Adelaxie

mpjf01, 2016-12-15:
I thought a repository might be useful for discussions and suggestions for improvement. Such discussions could possibly give FRSKY a more balanced view of the worth or otherwise of suggested improvements/features. People sending emails to your tech support department would not do that.

Adelaxie, 2016-12-15:
Actually,we have a private discussion repository when in developping stage. Will add you in.
But for production , we can not set a repository for every products.