FrSky-OS/HORUS-X10-X10S-X12S-bug-report #39: 1.2.22 Glider menu

EngelMT, 2016-12-05:
In Glider menu the Motor does not work.

For excample:
Input: Motor Slider S2 or L1 or what you want
Output: CH7 Motor
Motor: ACTIV
Servo Monitor shows CH7 do not work. Position is -100.

Adelaxie, 2016-12-06:
Please set SAFE SW and START SW under MOTOR page to ON then the motor source will work.

EngelMT, 2016-12-06:
Adela, you are right, SW set and it works. Little bit overdrawn but o.k.

EngelMT, 2016-12-06:
O.k. it is possible to set SAFE and START switch with the same switch. Make it easier. But customers ask us about to use also L1, L2, S1, S2, LS and LR for it.

Same is in TIMER. For start Timer it is only possible to use switches but not the sliders or potis.

EngelMT, 2016-12-08:
For better explain. Customers will use the slider or poti in combination. They will start the timer and in same time the motor with the same slider or poti.

Adelaxie, 2016-12-13:
You can set the potentiometer as Logic switch ,set the Logic sw as the control source of SAFE and START switch.
If you still have suggestion on this menu,please set a new issue.

closed 2016-12-13 by Adelaxie