FrSky-OS/HORUS-X10-X10S-X12S-bug-report #42: CURVE LIB

EngelMT, 2016-12-07:
CURVE LIB / Multi Line. Field for "FILE NAME" to small. Same in English and German.

EngelMT, 2016-12-07:
If you have a look to DR/EXPO you will find out that it is right here. So it seams to be a mistake in CURVE LIB.

Adelaxie, 2016-12-15:
#23  closed ,dumplicate post

Adelaxie, 2016-12-15:
>Field for "FILE NAME" is to smal. But bellow th numbers are normal the NAME of the file.But if I change the Number for excample from 11 to 3. Then I can see in 21-point line that the number off points are changed from 11 to 3. So it is not the name, but the quantity of points I can set.

Donot understand what you mean.

>

Adelaxie, 2017-11-23:
out of date issue?

EngelMT, 2017-11-23:
Can be closed. Field is bigger now and works in english and for German I found a shortcut.

Only for your information. 1.4.04 cut letters from WL TRANSMISSION.

Adelaxie, 2017-11-24:
Thanks and you can close issues set by yourselves actually.
WL Trainer page you mean I guess , will check.

closed 2017-11-24 by Adelaxie