FrSky-OS/HORUS-X10-X10S-X12S-bug-report #43: PITCH-CURVE (Airplane)

EngelMT, 2016-12-07:
Can everybody explain why we have a PITCH-CURVE in Airplane section? I think this can go away.

joel-1, 2016-12-07:
This is the same as your issue #26. Probably should close one. I don't know if it was the intent, but you can get variable pitch props for model airplanes ( at least you used to be able to ).

EngelMT, 2016-12-08:
You are right. Same as #26. I will close it.

I must smile a little bit. You are right we can use it for variable pitch props. O.k. so we will do it for a extrem smal quantity of model builders. Much more important will be to get the flight modes in Airplane.

joel-1, 2016-12-09:
Ha, yes, I only came as close to thinking about a variable pitch prop once.

Adelaxie, 2016-12-13:
@EngelMT  if we can close this issue now?