FrSky-OS/HORUS-X10-X10S-X12S-bug-report #44: CURVE-LIB Export/Import

EngelMT, 2016-12-08:
Has anyone already tested the export and import? I always get error messages. What am I doing wrong or does it really not work?

Adelaxie, 2016-12-13:
which version do you use? we did not reproduce it.

EngelMT, 2016-12-13:
I tested it in the moment with 1.2.23LTemp2 1212. Is it the same. Can be a mistake on my side. Please explain how to do export. SD Card is inside. Do I have to create a special folder?

[image]

Adelaxie, 2016-12-15:
#23 closed ,dumplicate post.

Adelaxie, 2016-12-15:
It is better to take a video of this issue, it works properly for us .

EngelMT, 2016-12-15:
I do not know wath a video can help but I will send it by skype. If it is my mistake than please expain how to use Export and Import.

Adelaxie, 2016-12-16:
You must pressed RTN to cancel the name editing,so the name can not be blank ,the error message pop up.

EngelMT, 2016-12-16:
I do it. In the moment I press the RTN the message pop up.

EngelMT, 2016-12-31:
Operator error. so I close it.

closed 2016-12-31 by EngelMT