FrSky-OS/HORUS-X10-X10S-X12S-bug-report #47: Translation char Č at begin of string

Fires04, 2016-12-10:
Hi, I maked translation FrskyOS to Czech language. But I found bug. If is the first char in string = Č - czech char, that char is not displayed in OS. If the char Č is somwhere in string it is no problem. 

[image]
[image]

janekx, 2016-12-12:
Know issue, will be fixed in new public FW with localisation

janekx, 2016-12-12:
bad button

Adelaxie, 2016-12-12:
For now, we think it is caused by .csv file generated by different kinds of softwares.
We engineer likes to test with txt editor and always has no issue.
So which software do you use?
can you try to use txt editor to make sure of it as well?

Fires04, 2016-12-12:
Hi,  i used google docs.   I will try different editor.

Adelaxie, 2016-12-13:
Please test latest firmware here.
http://221.228.240.132:8001/Adela/.horus_localisation1212.zip

Note:Please remove the _1212 in the tail of the file name.

Adelaxie, 2016-12-28:
L version fixed the issue .

closed 2016-12-28 by Adelaxie