FrSky-OS/HORUS-X10-X10S-X12S-bug-report #48: [1.2.23] New Binding Procedure

mpjf01, 2016-12-11:
There is a popup now with the binding procedure in D16 Mode. This offers the options

WITHOUT TELE 1-8CH
WITHOUT TELE 9-16CH
WITH TELE 1-8CH
WITH TELE 9-16CH

What's happened to the normal "with tele 1-16CH" that we all use?

This popup implies that the new binding system can be activated for all X series receivers. AFAIK it is currently restricted to X8, X6 and XSR and the new XM series. The availability of the "state" suggest that the selected option has actually been implemented, presumably therefore it is planned to release firmware for all X series receivers that supports this. Is this so?

Once binding is finished and you move to normal (or range ) mode the popup vanishes so you can't see what you have done without accessing bind again. It would be good to be able to see this at any time.

In my case the binding process froze with the banner WITHOUT TELE 1-8CH across the screen and the radio not responsive to any input ( the cursor moved around the screen under the banner but did nothing, keys unresponsive. had to power off.

joel-1, 2016-12-11:
I don't think this is a problem. With XJT module and an Xxx receiver this assigns which channels show up on the receivers PWM connectors. In the case of the X8R ( does the max 8 channels on the connectors ), selecting 1-8 will have the receiver connectors be channels 1 - 8,  Selecting 9-16 will have the receiver connectors be 9-16, all channels are available on SBUS. This should be what's going on.

Adelaxie, 2016-12-12:
@mpjf01 
>Once binding is finished and you move to normal (or range ) mode the popup vanishes so you can't see what you have done without accessing bind again. It would be good to be able to see this at any time.

Good idea.

>As a test, I bound a X8R and a X6R (latest firmware) and a S6R and a XSR (beta firmware) in the 4 Modes and could detect no difference, all cases appeared to be bound as telemetry, Ch 1-x.

XSR does not work as well?

>The new pop-up does not work with an external XJT module, is this intended?

Yes,it wont work. Only X12S has this function.

mpjf01, 2016-12-12:
I upgraded one of my XSR receivers to the newly released beta firmware and could see no change, all 4 bind modes gave telemetry data and channels 1-x. It may be important to note that I have so far only tested with PPM output, I will try SBUS later tonight.

Please note the comment that looking to see how a Rx has been bound seems to set the value back to "WITHOUT TELE 1-8CH" and the receiver unbinds.

mpjf01, 2016-12-12:
Have now tested SBUS with XSR. Same as with PPM above.

janekx, 2016-12-12:
I have tested XSR with beta and telemtry was not transmited when was bind in non telelmtry mode.

mpjf01, 2016-12-12:
More information please.

Taking account of the comment above I reflashed one of my XSR with firmware downloaded from this link
http://frsky-rc.com/download/view.php?sort=&down=313&file=Beta%20Firmware-XSR
- same as yesterday. I get telemetry and Ch 1-x in all 4 bind modes. I am using a beta Horus with FCC firmware.

Adelaxie, 2016-12-13:
Please provide your test procedure.
http://www.frsky-rc.com/download/view.php?sort=&down=313&file=Beta Firmware-XSR
This firmware have been proven by many users and testers.

closed 2017-11-23 by Adelaxie