FrSky-OS/HORUS-X10-X10S-X12S-bug-report #49: New Sticks with switches

EngelMT, 2016-12-12:
After mount the new sticks with 2-pos switches (connected on the LS and RS connector) I found out a strange behavior. Both switches can be used only from 0-100%. This is probably due to the 2-pos switch.

But when you press switch 1, you can see on the servo monitor that the path of the channel is only 78%. When you press the switch 2, the path goes to 100% and switch 2 to -70%.

Switch 2 seems to be only -70%. Whether it is operated alone or with switch 1.

In addition with the problem that the push button switches can not used in the moment this new sticks are not possible to sell for the Horus. FrSky has to work hard and fast to solve it.

Adelaxie, 2016-12-13:
Will test on it.

EngelMT, 2016-12-13:
Will be good if you test it self. Because my only further help can be to make a video. But I know from several customers that this problem is true.

EngelMT, 2016-12-13:
By the way. Customers and we too think that the very small screw is very bad. Better to use M3 screws.