FrSky-OS/HORUS-X10-X10S-X12S-bug-report #53: 1.2.23 Glider menu

EngelMT, 2016-12-13:
As I wrote in the other issue customers will use a poti like S2 to start and adjust the motor and in with the same poti they will start the timer.

Adela wrote to use the Logis Switches. But then I get only the function of a switch and not of a poti. And the timer will not start. Can be that I make mistakes in settings.

So please explain step by step what the customer has to do. If it works it is no issue. But I am skeptical that it works.

Adelaxie, 2016-12-15:
1. In SPEC FUNC, source1 select S1,source2 select const such as 20%. option selcect > or <.
2. You can check the LSW state in the third main page.
3. Se the Swtich in MOTOR page LSW.

of course ,LSW is still a value of switch ,the output can be only 0 or 1 that generated by the S1 and the option you set.

EngelMT, 2016-12-15:
I do it like you wrote. Motor works but timer can not be stoped if the motor is stoped. I check 20% and several other ones. But never the timer can be stoped by S1. Do you have an idea?

Adelaxie, 2016-12-16:
Also start the timer need you set the LSW in the timer START position.
I missed your demand of this.

EngelMT, 2016-12-16:
[image]