FrSky-OS/HORUS-X10-X10S-X12S-bug-report #57: Variometer problems

soarersteve, 2016-12-13:
Flew the Radian again today with the upgraded sound pack. Unfortunately there are still problems.
I had maximum altitude (maxalti-v) and altitiude (alti-v) showing on the Horus display and also both parameters linked to a special function switch to play value. (Both special function boxes ticked).
Both Maxalti-v and Alti-v displayed correctly on the Horus screen.
When the switch was thrown to announce the current altitude it was accurate until I went over 101.7 metres. After that it continued to announce 101.7 metres despite the display showing the correct altitude.
The announcement for Maximum altitude only ever said zero metres despite the display showing the correct value.

Adelaxie, 2016-12-15:
On which version of softwares please?

soarersteve, 2016-12-15:
Software version 1.2.21

Regards
Steve

Adelaxie, 2016-12-15:
@soarersteve  Please try on latest v1.2.23 in FrSky website download section.

soarersteve, 2016-12-15:
Please send me the new software by mail and I will try this weekend.
Thanks.

Adelaxie, 2016-12-16:
@soarersteve  I did not find your mail address from your profile,so here is the software for test:
Caution: This firmware is not fully tested and only for this issue. Please be careful when flying.
[1216_frtx_test_firm.zip](https://github.com/FrSky-OS/X12S-bug-report/files/656177/1216_frtx_test_firm.zip)

soarersteve, 2016-12-17:
Adela

Thanks for the file.

I see there is no mode1 / mode2 option for the firmware files.

As I am a mode1 flyer will this make any difference?

Regards

Steve 

Sent from Windows Mail

From: Adela
Sent: ‎Friday‎, ‎16‎ ‎December‎ ‎2016 ‎01‎:‎11
To: FrSky-OS/X12S-bug-report
Cc: soarersteve, Mention

@soarersteve I did not find your mail address from your profile,so here is the software for test:
Caution: This firmware is not fully tested and only for this issue. Please be careful when flying.
1216_frtx_test_firm.zip

soarersteve, 2016-12-18:
Adelaxie
I flew again with the new software and both problems are now fixed!
Thanks for your help.

closed 2016-12-19 by Adelaxie