FrSky-OS/HORUS-X10-X10S-X12S-bug-report #58: Dangerous switching off the radio with plane in the air, when the battery is low

Woita2095, 2016-12-15:
I came across to a very dangerous feature:

When the battery inside Horus is discharged, the warning pop-up window appeares on the LCD.
The user have to answer, whether switch off the radio or not.
But the answer is ambiguous (I am not sure, but something like OK/Cancel). 
There is serious chance to unwanted switching the radio off, having aeroplane in the air.
In adition, manipulation with rotating knob (encoder) is difficult, when , and there is probability to skip to unwanted position, pressing Enter.
I STRONGLY SUGGEST:
Not switch radio via rotating knob and enter at all, but ONLY via main power switch.

The pop-up window should be only information, not action, and even switching the radio off!!!

janekx, 2016-12-15:
I agree,  that encoder is too weak to select and this message should not be so dangerous. 
Lets radio switch off when not enoight power but not by wrong or unwanted selected answer

Adelaxie, 2016-12-28:
There is an option in General page that you can choose to skip the warning when battery low.
How we change it like this:

1. like now
2. When you select to skip the warning,the warning on GUI will not pop-up,just voice warnings.

Woita2095, 2016-12-28:
ad1. - Very dangerous, as I have written above.

ad2. - If I choose "skip the warning", it should actually "skip the warning". But you suggest "just voice warnings". It is not clear solution.

The only propper solution is to remove the possibility to switch off the radio via rotating button. It means to leave the warning as is (both voice and LCD), but just remove the choice switching off from the LCD by manipulating with rotating button (remove the question about switching-off).

Adelaxie, 2016-12-28:
When the battery warning occurs, the low voltage may cause some part of the radio do not work properly even though you can see the screen works good .
So we recommend that users will choose to fly back and power off the radio other than waiting for the radio cut down by itself when you see the warning.
Your suggestion will be good considered ,thanks.

Woita2095, 2016-12-28:
Of course, but unwished switching-off while flying will cause imediatelly crash the plane. The only right procedure for pilot is to land and then manually switch transmitter via the main power button, not to solve switching-off by ambiguous manipulation with rotating operating element, with plane in the air.

MIKE-DAILEY, 2016-12-28:
I've been flying X2S systems since March and I do not see this as a problem.

tomkfly, 2016-12-28:
I would hope to never fly with a Tx battery anywhere near a cut off voltage, or even a low battery warning, but I still think auto shutdown is a no no. I don't believe any other transmitters have such a function.

Woita2095, 2016-12-28:
@MIKE-DAILEY "I've been flying X2S systems since March and I do not see this as a problem."

OK, it is not everyday event, but if once will happen, it will be really stressful situation. You must decide which option to choose, and with the rotating button it is like Russian roulette. Could you imagine such a situation?

Woita2095, 2016-12-28:
There are two screens. 
The first is : [http://foto.ludl.cz/Dangerous.jpg](url)
The second is: [http://foto.ludl.cz/Correct.jpg](url)

**... and the very simple solution is**: To leave out the first screen and use only the second screen. It is easy for programmers and it is a good solution.

GliderMisguider, 2017-03-03:
Is this issue a software issue only, or is there a hard limit for low voltage?
If the latter (= hard limit for low voltage) what is that voltage?

liukku56, 2018-01-29:
There must be no other way to turn off the radio than via the main power switch. Everything else is not acceptable.
I have accidentally switched off the radio by mistake when the warning signal was displayed, luckily, it was at the desk.
This must get the highest prority.

/Ville

Adelaxie, 2018-01-30:
@liukku56  Please disable the warning page in gerneral page under MODEL menu.

liukku56, 2018-01-30:
>@liukku56 Please disable the low battery warning option in general page under MODEL menu.
Adelexie, I am not happy with that solution!

/Ville

Adelaxie, 2018-01-30:
Just an idea may make it better:
How to skip the warning as default and user can enable it by setting in the gerneral page?

Adelaxie, 2018-02-02:
Please try this firmware

MIKE-DAILEY, 2018-02-02:
1.5.03 is working well for me with testing battery low alarm on my X12S.  For anyone that wants to test it, be aware that if you up date to it do not try going back to 1.4.04 without following the steps in the compatibility document here https://www.rcgroups.com/forums/showpost.php?p=35634343&postcount=950  Before updating to 1.5.03 (only for X12S) make a back up of your flash memory drive.

choochoo2, 2018-02-05:
I agree with liukku56 on this, there should be no other way to shut off the tx from the firmware.  The tx should remain transmitting until the switch is pressed or the hardware fails from low voltage.  

My X10 also gave me a low voltage warning, I also accidentally confirmed shutdown.  Fortunately I was also not flying at the time but my first thought was; wow, that would have been the end of the airplane, at best.

If this has happened to two people with the X10 still so new it is inevitable that it will cause a crash.