FrSky-OS/HORUS-X10-X10S-X12S-bug-report #59: LSW madness

janekx, 2016-12-16:
There is many issues in LSW. 

1. the LSW cant be switched if we compare channel to constant : 
LSW1 Ch7 > constant +60 this condition not work if are used >< =

2.  the LSW cant be switched if we compare channel to pot 
LSW1 CH7> pot S2 
S2 pot satisfy condition only in 50-100 % of own range and not -100 to 100 

3. why have constant 5 zerows and somewhere 3 or 4 based on pot, SW, CH, LSW ?
  

4. If i think switch like source of signal -100 - 100 there is still not possible compare with contant
LSW1 SW F up < consta 40

This are basics that i why i mark critical

EngelMT, 2016-12-19:
Yes you are right. Wrote it to Adely by skype because customer ask us about.

Adelaxie, 2016-12-20:
fixing it.

Adelaxie, 2017-11-20:
Please check new firmware

closed 2017-11-20 by Adelaxie