FrSky-OS/HORUS-X10-X10S-X12S-bug-report #60: Volume problem

soarersteve, 2016-12-18:
When assigning the volume to a slider or pot approximately the first one third of the movement does nothing then the volume jumps from zero to too loud.
Could the volume adjustment be made more linear please.

EngelMT, 2016-12-31:
You are rigth. Here a video to show this problem.
[MAQ01713.zip](https://github.com/FrSky-OS/X12S-bug-report/files/679572/MAQ01713.zip)

FordT, 2017-04-06:
Have there been any progress on this? Most every pot on my Horus has the same problem, some times it will be fine from 0 to 50%, but from 51% to 100% it will stay the same as 50% all the way up to 100% or visa-versa.

Adelaxie, 2017-11-23:
The pots have reserve zone (dead zone ) in the two end. Now , we have made full use of the function part.