FrSky-OS/HORUS-X10-X10S-X12S-bug-report #61: Internal Memory

EngelMT, 2016-12-19:
Now I get first responce from customers who told me that the internal memory is not big enough.

A customer now has 20 models in memory. All with photos and with names as language announcements. It only has about 3 MB of space on the memory.

Why do you have sounds and images on the internal memory. These are not safety-relevant and can be on the SD card!

MIKE-DAILEY, 2016-12-19:
You know that the sounds can be on the Flash Drive and the SD card?  If an individual sound is not on the Flash Drive it will look for it on the SD card.

EngelMT, 2016-12-20:
O.k. thanks for the info. We will check it. But Images?

Adelaxie, 2016-12-21:
1. FIRMWARES/LOGS/MUSIC are always saved on the MicroSD card to create more space on the
internal flash memory.
2. SOUNDS/en/ Tracks   and SOUNDS/en/Models can be put under sd card too. these 2 folder contents can be added by users.
the other 2 SOUNDS/en/System and SOUNDS/en/Value can only be put in internal flash as these 2 folder has the default files that can not be expanded by users.

EngelMT, 2016-12-21:
OK. We will see until the first customer logs in because the internal memory is full.

Is it such a big problem to change the software, all the sounds and images are stored on the SD card? I mean anyway these things have nothing to look at the internal memory. This is not a safety-relevant data.

I can remember that we have already discussed the issue a long time ago during the beta test phase. Actually, everyone was of the opinion that these files would have to be on the SD card. Only FrSky disagreed.

What happens if the internal memory is full? Has anyone already tested it?

Adelaxie, 2016-12-22:
You will get error message: "SD card not exsit" before and "SD card error in future".

EngelMT, 2016-12-22:
O.k. important is that it is tested and it is impossible to crash the radio. O.k. than we hope that the memory is big enough.

closed 2016-12-22 by EngelMT