FrSky-OS/HORUS-X10-X10S-X12S-bug-report #69: Glider Trim

EngelMT, 2017-01-02:
In glider mode, (I think in other too) it is important to be able to trim differently in all flight phases. That does not work. The trim values apply to all flight modes. That has to be changed.

What is even worse is that the trim values have disappeared after the transmitter is switched off and on. That is, the values are not accepted. Only in Launch it is saved.

But only if the switch for the Flight Modes are set like I wrote in Issue #68. Means first position of swich must be for launch and can not be used by speed.

Adelaxie, 2017-01-13:
Please set seperate in the TRIM SETUP page.

EngelMT, 2017-01-13:
What do you mean with "Please set seperate in the TRIM SETUP page."?

ycchu8, 2017-02-09:
I use FrSKY OS. I have tried "Separate" in Trim Set Up.  All trim memories in all flight modes disappeared after reboot.  Same issue in "Glider", "Heli" ... setting.

However, in "Combine" setting trim memory is retained in "Launch" mode (glider type setting).  I do discovered a trick. Upon reboot (do not touch any trim tab in this stage) switch to "Launch" mode and move any trim (one click would do the trick) and the trim settings will appear in all other flight modes.  This is definitely a FW issue.  FrSKY should fix this very basis problem ASAP.

Adelaxie, 2017-02-13:
@ycchu8   Please leave your mail address or send a mail to libo.xie@frsky-rc.com and I will send to you a test firmware fixing this. 
After testing,we will fix it in latest software. Thanks.

ycchu8, 2017-02-13:
Adelaxie,
Please send Test FW to chuyatchor@gmail.com.  Thanks.

Adelaxie, 2017-02-14:
@ycchu8 
Sent and thanks, looking forward to your feedback.