FrSky-OS/HORUS-X10-X10S-X12S-bug-report #71: 1.2.25L CAMPER FLP->ELE MIX

EngelMT, 2017-01-06:
Has anyone of you tested this mixer in airplane mode? Either I make a mistake or the mixer has no function at all.

Adelaxie, 2017-01-13:
Please check the CAMB in input map. Is your wrong understanding of CAMBER ?
Same to #71

EngelMT, 2017-01-13:
Adela, please explain me what I understand wrong? Several customers have the same problem. I think there are two ways. The software has a mistake as I wrote or your manual is bad?

I think I know what CAMBER is meant. This causes all the flaps on the wing to be lowered to increase the buoyancy. The airfoild is more vaulted.

I do not understand why the CH 15 does not run even though this is assigned to the CAMBER encoder by the software. Of course I can change input and output map. But the CH off CAMBER do not work every time.

So please explain the function of the CAMBER MIX and how to programm it!

Adelaxie, 2017-11-23:
Please test if it still exsist now?

EngelMT, 2017-11-23:
I think it is solved. No customer inform us about problems last time.

Adelaxie, 2017-11-24:
Thanks for confirmation.

closed 2017-11-24 by Adelaxie