FrSky-OS/HORUS-X10-X10S-X12S-bug-report #73: Sound Alarm too loud, Poti Problem?

EngelMT, 2017-01-07:
Customers informed us that it is not possible to make the volume for the voice announcement quiet enough. I have tried it and must say that the customers are right. With earphones it is particularly annoying.

Can that be at the Potis? Because they have a range of about 10-15 degrees where they do not show any effect. This is also a big disadvantage when the potentiators are used for functions such as flaps etc.

Adelaxie, 2017-11-20:
And now?

EngelMT, 2017-11-20:
Seems to be better with 1.4.04. Possible to adjust over the complete way of 360 degrees of the poti. So I think you can close this issue.

Sorry, but I did not pursue Git Hub, because nothing happened there.

Adelaxie, 2017-11-21:
It is OK and i will close it then.thanks

closed 2017-11-21 by Adelaxie