FrSky-OS/HORUS-X10-X10S-X12S-bug-report #74: Needless adjustment possibilities

EngelMT, 2017-01-08:
In many menus we have in page 1 and page 2 the same functions. This is usually the status and the associated switches. Changes on page 2 will automatically affect page 1 and vice versa. This is completely unnecessary and confused only. As an example, I call the menu "Snap Roll". But many others are also concerned.

Adelaxie, 2017-01-13:
But do not know how to improve....

EngelMT, 2017-01-13:
I think your programmer can do it. He has to look the menus with more than one page. And then he must look where there are control panels, which already exist on the previous page and have the same function.

If he has found this, he can delete the superfluous control panels.

I am aware that this is also a programming effort. Such changes I would always synonymous separate. So a special update which only the testers get. Only when they after long tests say it is o.k. such changes should be incorporated into the normal updates.

Because there is the danger that one with such a change again other mistakes.