FrSky-OS/HORUS-X10-X10S-X12S-bug-report #76: EXTRA MIX

EngelMT, 2017-01-09:
I use STCK. Mode 1. Means THR on Stick J2. If I move this stick in EXTRA MIX to set SRC it comes ELE (J3). Seems to be not the right stick mode. But I have set it right. If I check Input and Output and the Servo Monitor I see that Stick J2 it THR.

EngelMT, 2017-01-11:
To get the confusion completely the following test.

Now I change my Config Files to the files who are added in the update folder from FrSky. Now the setting in the Input Map is THR and J3!!.

But if I move the Stick J2 now in Extra Mix than the setting in SRC will be also ELE and J2 like as wrote above!!!

O.k. I change the setting in Input Map ELE and J3 and THR and J2. In Extra Mix will be the same problem. Move the right Stick J2 and it will be set ELE J3.

It is not normal, or?!?!?

WismyYao, 2019-05-31:
Fixed problem

closed 2019-05-31 by WismyYao