FrSky-OS/HORUS-X10-X10S-X12S-bug-report #77: 1.2.25L SYSTEM / Model SELECT / CREATE

EngelMT, 2017-01-10:
If you create a new model and cancel the setting by the CANCEL Button or by RTN you get a READ ERROR!

MIKE-DAILEY, 2017-01-10:
Tested and not seeing on my production system with FCC 1.2.25.  Must be related to L.

EngelMT, 2017-01-10:
Do you have tested with a beta test transmitter or with a one out of production? I test it in the moment with a beta test and 1.2.25 and it is the same problem as with the production type and 1.2.25L

MIKE-DAILEY, 2017-01-10:
I tested with a production unit and my beta unit using 1.2.25 from the FrSky website download page and it works fine.

EngelMT, 2017-01-10:
I check it again with my production unit and 1.2.25. It is the same. I will explane again what I do.

-create a new model in mode Airplane
-Stick Mode 1
-Check the input. It shows THR with J2. It is o.k.
-Open EX Mixer, scroll to SRC and press "Enter Button"
-Move the right stick (J2)
-The setting is ELE and J3

I check it with this production unit and the 1.2.25 and 1.2.25 EU and now with the 1.2.25 NON EU. all is the same.

Now I create an airplane with Stick Mode 2 and do the same. Now it shows ELE but J2!!!!

I am sure that here is a fault!!

EngelMT, 2017-01-10:
To get the confusion completely the following test.

Now I change my Config Files to the files who are added in the update folder from FrSky. Now the setting in the Input Map is THR and J3!!.

But if I move the Stick J2 now in Extra Mix than the setting in SRC will be also ELE and J2 like as wrote above!!!

Adelaxie, 2017-11-23:
Out of date issues?

EngelMT, 2017-11-23:
I think this problem is solved in the last versions.

Adelaxie, 2017-11-24:
Thanks for confirmation of this issue.

closed 2017-11-24 by Adelaxie