FrSky-OS/HORUS-X10-X10S-X12S-bug-report #78: 1.2.25 Model Name

EngelMT, 2017-01-10:
If the model name is longer than you get an error if you copy a plane. It is logical because the horus will add "copy" to the name.

Better will be to add for excample a * in the front of the name if it is a copy. So the chance to get an error is less.

EngelMT, 2017-01-13:
For example:

Model name is "MX-2 Elektro". This are 12 letters or spaces. If you will make a copy of it the Software will add "copy" to this name. But it is not possible. It comes the warning you can see in the screen shot.

If you use a "*" instead the word "copy" you have more space for the model name. And this "*" is typical sign in programming.
[image]

Adelaxie, 2017-11-20:
Fixed with _ in the start of the name of copied file

closed 2017-11-20 by Adelaxie