FrSky-OS/HORUS-X10-X10S-X12S-bug-report #79: 1.2.25 Helicopter Flight Mode "HOLD"

EngelMT, 2017-01-15:
I know that this function is an idea off Futaba. But it do not work like on a Futaba transmitter I think.

Idea is to get with "HOLD activ" trottle in low rpm. Idea is to have no risk if you will do settings in other flight modes and test it on the Heli live. RPM everytime down. Regardless of the position of the throttle stick.

In Horus the THR-Hold must be activ in Flight Mode "HOLD". It is o.k. But now it must be possible to make settings in for excample "IDLE 1" and "PIT CURVE". You can do it but it will not work if Flight Mode "HOLD" is activ. So you can not check the right setting of "PIT CURVE".

Thus, this function brings no advantages. I am not a Helicopter specialist. So will be good to get comments from Heli specialists!

joel-1, 2017-02-08:
When I set up my helicopters, I don't think of a flight mode "Hold". Instead I think of an Auto ( auto-rotation ) mode that has a pitch curve that is usually different than the other flight modes. It will have negative pitch set for optimum auto-rotation performance and maximum set to max the linkage allows. The auto mode will also have a minimum throttle setting that is slightly above the "throttle hold" setting for a more reliable idle while practicing auto-rotations. Throttle hold should be able to be active in all fight modes. I have not checked this in FrOS because so far I only fly the electric helis with my Horus.

Adelaxie, 2017-02-10:
These all can be done now .