FrSky-OS/HORUS-X10-X10S-X12S-bug-report #82: Incremental encoder

EngelMT, 2017-01-18:
Customers inform us about problems with the Incremental encoder. Means black button for scroll and enter.

Button is broken by several customers. But more important is the problem to find the right position by scrolling before press enter. I think FrSky mus change the resolution per 360 degree turn.

Or the mechanic

Adelaxie, 2017-01-22:
We have replaced the plastic button part into new materials.
For the wrong operation ,will try to make it better.

EngelMT, 2017-01-22:
Alexa, please have a look to the Q X7. Ther the incremental encoder works perfect I think. So it must be possible to make modifications on Horus. Best was will be to do it by software. Then all customers can benefit from it.

Adelaxie, 2017-11-23:
Issues out of date

closed 2017-11-23 by Adelaxie