FrSky-OS/HORUS-X10-X10S-X12S-bug-report #95: Stick Mode

EngelMT, 2017-02-01:
If I had set a model for excample with stick mode 1 and another model with stick mode 2 it will be logical to safe this stick mode in the modell memory.

But if I change (select) to the model with stick mode 2 it works with stick mode 1!!! It is very cristical!

If we check the embroidery mode with each model before we create the model memory, then the selected mode must also be saved and activated when the model is called. Otherwise this is dangerous!

FrSky-GX, 2017-02-09:
>If I had set a model for excample with stick mode 1 and another model with stick mode 2 it will be logical to safe this stick mode in the modell memory.

It works like this ! and if  you create your model under CUSTOM mode ?
Please describe more about your issue.

EngelMT, 2017-02-10:
I tried to understand the problem. But now it functions as it should be. No idea what I did at that time. I found these problems while I tried for the German manual.

closed 2017-02-10 by EngelMT