FrSky-OS/HORUS-X10-X10S-X12S-bug-report #97: Input and Output Maps

EngelMT, 2017-02-01:
In Airplane, Helicopter and Glider all Output Maps are the same. No matter which embroidery mode was selected.

But in Glider V-Tail the Output is different in all Stick Modes. It is not logical.

FrSky-GX, 2017-02-09:
>But in Glider V-Tail the Output is different. Stick Mode 1, 3, 4 are the same. But Stick Mode 2 is different. It is not logical.

OutputMap is different based on different mode under v-tail glider? NO! they are still the same only input map is different.

EngelMT, 2017-02-09:
Here are Screen Shots. Firmware 1.2.25L

Mode1 Input
[image]
[image]

Mode1 Output
[image]
[image]

Mode2 Input
[image]
[image]

Mode2 Output
[image]
[image]

Mode3 Input
[image]
[image]

Mode3 Output
[image]
[image]

Mode4 Input
[image]
[image]

Mode4 Output
[image]
[image]

EngelMT, 2017-02-09:
[image]

Adelaxie, 2017-02-10:
Do not understand!!
Let us see CH9 of OUTPUTMAP under mode 1 2 3 4.
all the mode the output is FLP3 in your screenshot.
Why in your form, you list KLP3(assume it is FLP3) under mode 1 3 4 and QR3 under mode2  ???

EngelMT, 2017-02-10:
I think was my mistake, sorry. Forgott all. I check it and you are right. It works.

closed 2017-02-10 by EngelMT