FrSky-OS/HORUS-X10-X10S-X12S-bug-report #99: FrSky-OS,Horus cross-trim option

firstfishon, 2017-02-15:
I'm a new owner of the Horus running FrSky-OS and would like to see if someone can create an option to re-assign the trim switches of the ail to rudd and ele to throttle.

Allendaheng, 2017-02-16:
sorry, can you re-explain “the ail to rudd and ele to throttle.”?it is not easy to understand for other people...

firstfishon, 2017-02-16:
Hi Allen, I'd like to be able to have my aileron trim switch swap'd out with the rudder trim switch and the same with my elevator trim being swap'd out with the throttle trim switch, that way (with mode 2) I can use my left hand to trim my elevator and aileron with out letting go of my right stick.  This was easily selectable with my old 9xr-pro using the oem firmware, its super convenient, much easier. Thank You for you and your teams hard work.

Allendaheng, 2017-02-16:
Hi,Firstfishon sir
At now, X12S doesnot support the action.
But to be honest  ,your idea is very intersting.
I will try to suggest this to team and give you the feedback .
Thank you.

goryc, 2017-05-14:
I would also like to add my report; Yesterday at the camp I wanted to try at my friend a model and I exchanged J4 with J1 (from Mode 3 to Mode 1) but I could not exchange the trim. This was very serious for me! Switch from Mode 3 to Mode 1 and vice versa and Mode 2 to Mode 4 and vice versa should always be possible! Sorry for my intervention. On Futaba T18MZ you can normally make all possible exchanges!

F4Stan, 2017-12-19:
I also like to add a suggestion/ explanation.
In the menu TRIM SETUP you could add the option to assign trims to inputs

or

make it selectable on the INPUT MAP page.
For example input J4, trim selectable in a drop down menu:  "None, T1, T2...-T6"

This will give the user the flexibility to assign trims to inputs as he likes.
Often on electric gliders the engine is controlled by a switch and not the throttle. so throttle trim could be used for something else,

DutchRC, 2018-01-12:
I'm actually a bit surpriced cross-trim isn't allready in there :) 
(I've been looking for a way to do so for an evening)..  

Up to now I fly all my airplanes with a Futaba T14SG in which I allways use cross-trims.. Once you're used to it it's hard to do without :) 

To be honnest, I don't really care all that much How it's implemented. Model based would work well.. if it has to be a global setting I'd be happy with it as well..
Here is how it's done on the Futaba T14SG, in the inputs setup: 
https://4.bp.blogspot.com/-eJg9e01YgZ8/UeRHnPig0hI/AAAAAAAABB4/_PyRz2UVGpo/s1600/T14SG-linkage-function-1_v1.jpg 
Where I would then have J1 & T4 for the ailerons for instance..

mccolld1, 2018-06-07:
I would also like a way to assign each trim tab to a control surface.  It would be best to have it model based.  I suggest the input/output section of FrOS.
I have always had this function available on my Futaba 9zap and really miss it.  It makes trimming a new craft or a way out of trim/balance craft easier and safer because it is unnecessary to release the main control stick (e.g. aileron & elevator) to make adjustments.
Thanks!