FrSky-OS/HORUS-X10-X10S-X12S-bug-report #102: Sound in 1.2.25 and 1.2.25L

EngelMT, 2017-02-27:
Customers inform us about problems with sounds. It works without problems some minutes. But then it stops and only the Haptic Modul works. After landing the transmitter must switch out. After switch on it works.

Last information from a customer is that he makes 6 flights last weekend. All flights are without telemetrie sound and tracking sound after some minutes.

EngelMT, 2017-03-04:
More exactly info to this issue. Customers tested now with 1.2.25 and after several flights with 1.2.25L. With 1.2.25 the telemetry contact lost one time in 10 flights. With 1.2.25L it lost 10 times in 8 flights.

So it is a problem of both software versions. But more critical with 1.2.25L.

Adelaxie, 2017-03-06:
@EngelMT  telemetry lost (all telemetry data not updated) or just sounds missing??

EngelMT, 2017-03-06:
Only sounds. Telemetrie ist well shown on display. And the haptic works also well. But sounds are lost and it is only possible to remove if the transmitter is switched out and on.

Adelaxie, 2017-03-07:
ok，thanks

Adelaxie, 2017-11-23:
OUT OF DATE ISSUES.

EngelMT, 2017-11-23:
Sorry, but problem is not solved. Some customers have this problem up to now.

Adelaxie, 2017-11-24:
really? I did not receive any of this kind , ok , reopen it and can you provide more information based on latest firmware please? thanks in advance.

EngelMT, 2017-11-24:
Yes really. Some days in every fligths and on other day only one time. Or never in some weeks and than again sometimes on one day. So it is not easy to find out the problem.

janekx, 2017-11-27:
SD card issue?

EngelMT, 2017-11-27:
Tested with severl new and different SD-cards.