FrSky-OS/HORUS-X10-X10S-X12S-bug-report #103: Wrong date in telemetry logs

trondgzi, 2017-03-02:
I have an FLVSS sensor and a FAS40S mounted in my plane and turned on telemetry logging. In my SD LOGS directory I found a file 02-30.csv and inside all dates are 2017-02-30.

The date and time on the screen is showing the correct date 2017-03-02

Adelaxie, 2017-03-03:
We did not reproduce it. Anyone has the same problem?

trondgzi, 2017-03-03:
I have attached the log (had to use .txt extension to upload) [02-30.txt](https://github.com/FrSky-OS/X12S-bug-report/files/816022/02-30.txt).

This was from the 2nd and 3rd flight with a fresh out of the box unit. All I had done to it was to create a few models, and update the firmware.

Adelaxie, 2017-03-06:
ok,got it. there is no day of 02-30, we tested with other days.
we need some special calculation when logging . Thanks for report.

Adelaxie, 2017-10-30:
old version issue.

closed 2017-10-30 by Adelaxie