FrSky-OS/HORUS-X10-X10S-X12S-bug-report #105: 1.3.01 GPS

EngelMT, 2017-03-22:
But it is not displayed on all display pages. In addition, the ads vary greatly. This can be a problem in the close range. At a bigger distance I have not tried it yet.

[image]
[image]

MIKE-DAILEY, 2018-03-01:
I do not have an aircraft GPS to test the X12S GPS function.  Does FrOS assume that the X12S is facing (pointing) north and plots the aircraft GPS location on the screen relative to the X12S that way?

EngelMT, 2018-03-02:
I can not tell you exactly. But I assume. The position of the model is still changing very much without moving with the transmitter or model. The distances are still very fluctuating. In my opinion, this does not work well with the FrSky OS 1.5.03.

In addition, it clearly takes too long until the GPS signal is detected and the coordinates are displayed.

MIKE-DAILEY, 2018-03-02:
Thanks for the info.  With the R9M in use and the IXJT off the X12S GPS finds the satellites pretty fast.

EngelMT, 2018-03-02:
Why do you think that it is a different between 2.4 and 900 MHz?

MIKE-DAILEY, 2018-03-02:
The 2.4 antenna being so close to the GPS module seems to cause the GPS to have problems finding satellites.   You can test this by turning off the IXJT and see how fast the GPS finds the satellites.   The R9M antenna position and the 900mhz does not seem to cause problems with the GPS.

EngelMT, 2018-03-02:
Have tested it in the moment. I can not feel any different. But another question. Do you have an idea why the GPS do not work in a customer transmitter? I change the GPS board and GPS antenna. But the same. This transmitter will not find the satellit.

MIKE-DAILEY, 2018-03-02:
With the IXJI turned off my systems find the satellites in less than 60 seconds.  

Not sure about your GPS problem.

Woita2095, 2018-03-02:
My X12S get GPS fix in 5 minutes (first start), 2 minutes (2nd start) and 1 minute (3rd start). Not matter if RF module is on or off. 
(A note: I use external antenna.)

EngelMT, 2018-03-04:
O.k. this is like my own test transmitter. But in customer transmitter it is different. O.k. I have to discuss it with Frsky.

brainbubblersbest, 2019-04-08:
Have The same Problem here. No GPS fix until i turn off internal tx. I need this to find my drone if there is crashed. GPS from drone over f. Port works fine. Internal x12s works like shit.
Thinked about a emv shield out of aluminium foil. 
Maybe Move the GPS antenna on other Position in the tx. At normal use, between x12s and the Sky. There is a thing Blocking GPS Signal named Display... Who the f...  Develope such a horrible Position for a GPS receiver antenna? 🙈