FrSky-OS/HORUS-X10-X10S-X12S-bug-report #106: 1.3.01 Alti-v and Alti-g does not acquire

MIKE-DAILEY, 2017-03-22:
1.3.01 Alti-v and Alti-g does not acquire.

n1973w, 2017-08-01:
also now in 1.4.01   :-).. You cant just brake code and not fix it for ever and ever..

MIKE-DAILEY, 2017-08-01:
I reported it when I was testing the pre-beta and again with 1.4.01.

strgaltdel, 2017-08-02:
Ich kehre zurück am 13.08.2017.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [FrSky-OS/X12S-bug-report]
1.3.01 Alti-v and Alti-g does not acquire (#106)" gesendet am 02.08.2017 01:09:18.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

strgaltdel, 2017-08-02:
Ich kehre zurück am 13.08.2017.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [FrSky-OS/X12S-bug-report]
1.3.01 Alti-v and Alti-g does not acquire (#106)" gesendet am 02.08.2017 01:53:31.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Nesan65, 2017-08-13:
is there any other way to work on veriometer ? why no solution yet

MIKE-DAILEY, 2017-08-13:
FrSky found what was causing the problem in 1.3.01 and 1.4.01 and are working on a solution.

Nesan65, 2017-08-15:
thanks Mike-Dailey the newer version 1.4.02 beta cure the veriometer problem!
Thank you Love Horus x12s.! and recommending to other co flyers.

MIKE-DAILEY, 2017-08-15:
Many thanks to the folks at FrSky that fixed the problem because it was complicated.

closed 2017-08-15 by MIKE-DAILEY