FrSky-OS/HORUS-X10-X10S-X12S-bug-report #107: 1.3.01. SW DN and SW UP

EngelMT, 2017-03-24:
Same as in 1.2.25. Switch up is if the switch is down.  And vice versa.

Adelaxie, 2017-11-23:
Different comprehension I think? did not get any silimiar report.

EngelMT, 2017-11-23:
However, Helmut Renz from Germany has written several times. The switch positions are exactly inverted in the Horus as in the other transmitters with OpenTX.

It is the question of whether one has to change that today, or whether everyone has gotten used to it.

The fact is that it is not logical if you move a switch up an ad gets down.

Adelaxie, 2017-11-24:
sorry I  did not remember the progress of the issue and Yes, I remember it now and R&D thought it does not influence the function so kept it as now.

EngelMT, 2017-11-24:
Right, it does not influence the function. But it confuses. It was noticed when you have brought the S-receiver on the market. If customers have a Taranis and a Horus, then the switch position is different for the Taranis than for the Horus for the same function. That has led to many discussions here.