FrSky-OS/HORUS-X10-X10S-X12S-bug-report #108: 1.3.01 S6R SETUP

EngelMT, 2017-03-24:
I think is not working.

EngelMT, 2017-03-24:
No it works after I test to copy the S6RConfig files in my Horus. I am missing an how to do!

MIKE-DAILEY, 2017-03-24:
Don't know if you follow the main X12S thread on RCG, but I covered a lot of how to do there https://www.rcgroups.com/forums/showpost.php?p=37153653&postcount=9179

Mike

EngelMT, 2017-03-24:
Hi Mike,
thanks for this info. But I think that it would be the right way if Frsky gives appropriate information with the update. FrSky can not expect customers to be informed in forums around the world.

A good documentation is missing at the moment at all products. Poor Adela is completely overworked in my eyes. FrSky must have personnel who do nothing else than write instructions and check software. And that must also be good model pilots to have the experience on what to look out for.

janekx, 2017-03-30:
I saw that there mus be cfg file copied in relase notes

n1973w, 2017-06-24:
When I tried this function, reciever had a hard time retaining data.. I had to navigate menus somehow in a very specific way, or it would just say "0" although i had just entered values.  In any case it was very hard to use..

Adelaxie, 2017-10-30:
old version issue.

closed 2017-10-30 by Adelaxie