FrSky-OS/HORUS-X10-X10S-X12S-bug-report #110: For reduse the wear of menu scroll button and fast nevigation.

Nesan65, 2017-04-07:
1st Assign reset button may help to reset entries and values  easier and faster this could reduce the wear of menu scroll button.
2nd May assign "SWH" two position not hold switch to  scroll lists and curve libraries even alphabets to continues scrolling to reach the value faster.
3rd Use Page Up Down to Jump from majuscule - minuscule to numbers and signs faster "as batch" so navigating to enter names and values in less effort

janekx, 2017-04-18:
1. duplicate of #92 
2. i not undeestand, can you better describe please ?
3. enhancement that was too allrerady discusessed and added to to do list with long press enter.

BadRaven, 2017-05-06:
Way Way back it was proposed and I thought accepted that the menu order would be swapped around so that most used options were all clustered together on the same screen. We still have the situation where you have to switch screens even between Input and Output which is very poor ergonomics. We still have the situation where little or even "once only during initial setup" options are presented on screen one forcing avoidable scroll wheel and screen changing.  This needs to be implemented before ANY other form of menu tuning!

MIKE-DAILEY, 2017-05-06:
Dave, I'm not sure when it changed because I'm running 1.3.01 but the screens have changed and input and output are together on the same screen.

BadRaven, 2017-05-06:
Cheers, I have been holding off the last couple due the uncertainties!

Nesan65, 2017-08-18:
we are using the menu button for scroll and enter so straining two much of it. how about if we can use Page up Page Down switch to jump for number of character for example 10th or 20th like. exa:- for naming a new model "Avistor" first letter Capital and have to roll long way to small v letter even for numbers. for enter 1045 why need to turn the menu button longer if we can jump by 100th with page button.

and how about assigning a reset button so can bring 0 or default value instantly.

Bravedan, 2017-08-20:
If we are now talking text entry:-

1. Short Press lower case, Long Press Upper case, which shortens alphabet scroll required.
2. Spaces between alphabet and numbers, not just one
3. Set should wrap, not reach end and stop.

Nesan65, 2017-08-20:
Bravedan's idea is good too!
I just bring nother idea "M-X ' M-Y" key  may be used to navigate. "menu and txt edit in default mode". and assign a switch for other use's.