FrSky-OS/HORUS-X10-X10S-X12S-bug-report #112: Special Function File Names scrambled

hmacdonald, 2017-05-07:
I use the special functions menu screens a lot to set up voice announcements to switches, etc, and they work just fine.
But, there is an issue. When I return to the special functions screens later, the file names in the “With” fields have all changed, every time, and with each release of FrSky OS right up to the present.

They still work as I set them up, but the file names are scrambled. This doesn’t mean I can’t fly, but it does mean I can’t easily go back into these screens and see what they are supposed to be doing - and it is worrisome.

Have you heard of this? What can I do to correct it?

Heyward

MIKE-DAILEY, 2017-05-07:
You are talking about the play track function in special functions?  How are your track sounds that you are using being created for your system?

hmacdonald, 2017-05-07:
Thanks for the reply. Yes the play track function.

I created some of the .wav files with Audacity, and some were sent to me from a fellow in England.

They all work just fine, but the file names get exchanged for the names of neighboring sound files.
It is a mystery.

Heyward

Heyward H. Macdonald, D.Min.
Charlottesville,  VA

Check on my book at:

https://www.createspace.com/5521578

MIKE-DAILEY, 2017-05-07:
Check your sound files and make sure they meet the X12S sound file requirements, especally the quitity limit.

WAV -TRACK FOLDER - Flash Memory Drive

 Size < 100K; Number <100; name length <10 chars (.wav not included); format should be .wav

 MP3 - MUSIC FOLDER - SD Card 

 Size not limited; number not limited; name needs to be in English; format should be .MP3. 32~320kbps

 Model Sounds

 size < 50k ,format should be .wav;, name length < 30 

 path: SOUNDS/en/model

hmacdonald, 2017-05-07:
They seem to be well within parameters.

I have attached URLs for two photos.  The first shows the .wav file names correctly as I entered them.  Then I went to the next Special Functions screen and back, and the file names had changed to what you see in the second url.  Interestingly, the incorrect file name tag comes in each case from 4 wav files further down in the list, but the radio functions properly, ie. sounds the correct announcement for the switch assigned. 

[image]
[image]

I am now going to print the wav files to see if there is some trash in there causing the displacement.
I will let you know what I discover, if anything.

heyward

hmacdonald, 2017-05-07:
Actually, as I look at the file list of wave sounds, the substitution is 4 files UP in the list, not down.  It seems to be down when using the controls on the radio.

MIKE-DAILEY, 2017-05-07:
How many total files are in the track folder?

hmacdonald, 2017-05-07:
> How many total files are in the track folder?

25 files.

Thanks for helping.

heyward

hmacdonald, 2017-05-07:
25 files are in the track file.

hmacdonald, 2017-05-07:
and BTW, no matter what computer I use, I can't get the USB flash and FrSky drives to eject.  I have to shut down my computers to safely remove the radio.  I don't know if this is related, probably not, but it is the other thing that bugs me.  My PC takes 15 minutes to reboot.

MIKE-DAILEY, 2017-05-07:
Is there one or more over the size limit?

MIKE-DAILEY, 2017-05-07:
Not ejecting is not good.

hmacdonald, 2017-05-07:
OOPS, I see one file in there from my friend in England that is way too large.  I have not tried to use it, 
Could that be causing the issue?  I shall remove it and find out!

h

hmacdonald, 2017-05-07:
OK, I have removed battchrg, batlow, high, packvlt, syson, and warbrd, so have 19 .wav files left in en.

The result is that the file names are now only 2 off from what they should be instead of 4, so we must be onto something,  I don't see anything else to remove that looks suspicious or that I am not likely to use.

Actually, I am not using batcrit, and thrdis, both of which came from my friend in England.  Could they be the issue?  I have included the new list of what is in the radio, below.

[image]

heyward

MIKE-DAILEY, 2017-05-07:
I do not recognize that screen.  Is it a Windows computer?

hmacdonald, 2017-05-07:
It is a Mac.  I got tired of the PC an hour ago and moved just to look at the files, not to flash anything.
but it is not the Mac's fault.  This issue has been going on for a year.

Also, the batcrit and the thrdis files are in use for another plane, and work fine.

So, I am down to 19 track .wav files, all of which work fine, yet their names are moved two down in the list in the track file.  It is 2 down, since th eMac sorted the file names for printing.

I am scratching myt head...

heyward

hmacdonald, 2017-05-07:
Wait, does the firmware pick up the track files from both the internal memory and from the micro memory card?  I see that some files are duplicated when dialing the radio wav files, but they are not duplicated in the track file of the radio.

MIKE-DAILEY, 2017-05-07:
Seems strange.  Might be that the file allocation table is messed up and that is why it will not close and eject. 

You might try deleting a couple more wav files just to see what you get at that point.  Then check the folder with the Windows File Manger to make sure there are only WAV files in the directory. 

If you create the same folder path names on the SD card you could put some of the files there.

Be carful with the Mac and the X12S because the Mac writes trash files to the X12S drives.

MIKE-DAILEY, 2017-05-07:
If you have duplicate files on the SD card that might cause problems.

hmacdonald, 2017-05-07:
OK, I have not used the Mac except to look at stuff, so will not do more than that.
I shall go back to my PC laptop.  I do hate it so...

My next idea is to clear the track files on the SC card and copy new ones from the radio.

I am going to keep at this.  Moving from 4 to three displacement is an indication that this problem can be solved.  I will let you know.

Thanks for the help.  If you get more ideas. let me know.

heyward

hmacdonald, 2017-05-07:
OK,
I will remove the sound files from the SD card and see what happens and will
let you know.

heyward
 
Heyward H. Macdonald, D.Min.
Charlottesville

Check on my book at:

https://www.createspace.com/5521578

On 5/7/17, 4:34 PM, "Jet_Flyer" <notifications@github.com> wrote:

> If you have duplicate files on the SD card that might cause problems.

hmacdonald, 2017-05-07:
OK Mike;

You did it!

I did not know that keeping a copy of the sound files on the SD card could mess things up.
I removed the wav files from the SD card and all the tags for the wav files on the radio suddenly were correct.

That is one of my two on-going issues.  Thanks.
The other, of course, is having to turn off the PC to eject the radio and SD card.  What a pain that is.  Evidently some process is going on that does not allow the USB to eject.  It is the radio since it happens on all three computers I have tried with it.  Oh well, that is for another day.  One victory is good though.

heyward

MIKE-DAILEY, 2017-05-07:
Good news! 

As I mentioned before you can have WAV files on the SD card and use them, but no duplicate names.  

The eject process should run and display a eject message that it is safe to disconnect, but the USB on the PC is still showen.

hmacdonald, 2017-05-07:
Got it, and thanks.

I was using the SDCard for back- up.
Nowhere in the documentation does it say I can't do that, so I am lucky to have you.

Heyward

Heyward H. Macdonald, D.Min.
Charlottesville

Check on my book at:

https://www.createspace.com/5521578

> On May 7, 2017, at 6:23 PM, Jet_Flyer <notifications@github.com> wrote:
> 
> As I mentioned before you can have WAV files on the SD card and use them, but no duplicate names

MIKE-DAILEY, 2017-05-08:
There is a lot of good information here in this document http://www.frsky-rc.com/download/view.php?sort=&down=377&file=How To-Horus X12S Flashing Instructions

hmacdonald, 2017-05-08:
Indeed so.
I do have that. Thanks for thinking of it, though.

Heyward

Heyward H. Macdonald, D.Min.
Charlottesville

Check on my book at:

https://www.createspace.com/5521578

> On May 7, 2017, at 8:01 PM, Jet_Flyer <notifications@github.com> wrote:
> 
> There is a lot of good information here

closed 2017-11-23 by Adelaxie