FrSky-OS/HORUS-X10-X10S-X12S-bug-report #113: 1.3.01 TELE CAL

EngelMT, 2017-05-19:
Can someone explain how this menu works? I can adjust what I want, it does not change anywhere.

There is also a mistake. Once you have made settings, they do not leave "zeros" anymore. The settings can be adjusted only to e.g. 0.01.

MIKE-DAILEY, 2017-05-19:
I noticed it will not go back to zero as well. 

The TELE CAL is used with the new 1.3.01 feature to use telemetry as an input.  As an example some folks want RSSI to be an input to a channel so their fight controller can receive RSSI on a channel and them display it on their video feed back data.   In that case TELE CAL lets you adjust how RSSI output changes the channel high and low.

EngelMT, 2017-05-19:
Hi Mike,
thanks for explain. But I used such feature without this menue TELE CAL. For excample reduce the power of an electric motor if the temperatur is over 85 degrees.

O.k. I will check next days how the TELE CAL works.

MIKE-DAILEY, 2017-05-19:
It is just used to adjust the span of the telemetry item if you are using it as an input.  In many cases it is helpful and in some cases not needed.

EngelMT, 2017-05-22:
O.k. thanks.

closed 2017-11-23 by Adelaxie