FrSky-OS/HORUS-X10-X10S-X12S-bug-report #114: 1.3.01 EU time and gps

xandrea, 2017-05-20:
in 1.3.01 EU version, with gps v2 sensor when fix the position the time of horus became incorrect drop 2 hours back...
The gps alt and position is not accurate and jump continuously do not fix a reasonable position...the alt indicate -13 metri......
I thinks is important to have a location  time menu  in horus for gmt zone time....and the menu for calibrate the gps telemetry...
Thanks and Sorry my english, write from italy...

janekx, 2017-06-01:
will look on the issue, you ask for time compensation against GPS time fot system time, right?

xandrea, 2017-06-01:
Thanks...yes the time of the radio when fix gps v2 sensor came back 2 hours and need manually adjust the time in  horus time menu..... but when fix the next time the sensor go back still....Need to adjust the gps time in horus menu  (different gmt zone) , without the sensor can modifay it indipendently.....----Messaggio originale----

Data: 1-giu-2017 8.25

A: "FrSky-OS/X12S-bug-report"<X12S-bug-report@noreply.github.com>

Cc: "xandrea"<xandrea@alice.it>, "Author"<author@noreply.github.com>

janekx, 2017-06-01:
OK lets need to add time zones set for GPS time tune. Thanks Andrea