FrSky-OS/HORUS-X10-X10S-X12S-bug-report #115: A joystick to navigate ...

goryc, 2017-05-22:
It would be very useful to use the joystick to navigate through the radio menus.
The X12S does not have a touchscreen LCD, so having a navigation system in the menu items would be very useful. You could use this feature only during editing within the menus; And you may be able to use it, with other custom features, while driving the models. What do you think about it?

janekx, 2017-06-01:
I think that joystick will be better to leave as the source because is useful for controling gimbals or so. FrSky  can consider the option fo controlimg menu, but what will it bring when there is on easy skipping cell to cell?

goryc, 2017-06-01:
When you are editing in all the menus (see Special Function), to move from one field to another you have to rotate the black wheel and to go back and forth you have to move from one field to another and this is very uncomfortable and makes Lose a lot of time; So I thought of a faster system to jump from one field to the other. I hope to have explained to me because of my bad English!

janekx, 2017-06-01:
Yes we can consider to find some way hot to skip by lines and not by cells easily, that will be usefull....

goryc, 2017-06-01:
Very Good